itead/ITEADLIB_Arduino_Nextion #6: Only one callback function

rastocs2, 2015-09-28:
in project i have 12 pages with cca 150 buttons where many buttons has same functionality
now i have to create many NexText and NexButton object and create callback function with parameter "void *ptr" for each object

maybe would be better have only one callback function with parameter "NexObject *ptr"

bspranger, 2015-09-28:
You can have a common function handle multiple buttons.  This is why there is the void pointer.

rastocs2, 2015-09-29:
is possible somehow check pid in callback function?

palivoda, 2017-01-04:
Hey folks, it's possible - overwrite notification event with your own code.  
1. In Nextion Editor select element and uncheck "Send Component ID" - this will disable event firing to arduino.
2. fire your own event, by adding command - "printh 65 01 01 00 ff ff ff"  so now you have event on page id 1 and element id 1. you can put any values of page ID and element ID    on this string. Nextion event handler will eat it and try to search for callback for this IDs. 
3. Define single callback function like 
NexTouch bSaveConfig = NexTouch(1, 1, ""); //this is hardcoded pageId and elementId in click events of all save config buttons

Enjoy.