itead/ITEADLIB_Arduino_Nextion #7: Functions for "set" uses only getObjName() for object identification

rastocs2, 2015-09-29:
Functions for "set" (SetText, SetValue, ...) uses only getObjName() for object identification, doesn't matter which pid is set:
e.g. i have 2 pages, p0 and p1
on both pages is text element t1
in arduino code is defined:
NexText txt0 = NexText(0, 1, "t1");
whet i use function txt1.setText(buffer), this function change txt value of component t1 on actual page
so when actual page in p1 then t1.txt on this page is changed
but txt0 was defined for page 0, not for page 1

bspranger, 2015-09-29:
Correct, but you should only update objects that are actually on the page.

In my fork I have made it possible to get page events as a callback.  Then in the callback you can push the data for that specific page.

Otherwise you had to always poll the display for it's current page.

bspranger, 2015-09-29:
Here is my code:
https://github.com/bspranger/ITEADLIB_Arduino_Nextion

DanNixon, 2015-10-30:
Does the instruction set actually allow you to change widgets that are on pages other than the current page without having to switch page?