itead/ITEADLIB_Arduino_Nextion #16: I can not compile

polyakovyevgeniy, 2016-08-03:
I downloaded the latest version: git clone https://github.com/itead/ITEADLIB_Arduino_Nextion and I copied it on the path C: \Program Files\Arduino\libraries, I then tried to create a project in the Arduino Ide and added #include "Nextion.h". but the project does not compile.

error:

`fatal error: Nextion.h: No such file or directory
 #include "Nextion.h"`

What am I doing wrong?

electrokean, 2016-08-03:
You shouldn't add libraries under your Arduino install folder.  Instead try putting the files under My Documents\Arduino\libraries\Nextion
This folder should contain all the Nex*.* files and should have the same name as the main library include file (Nextion.h)

bigredsmokehouse, 2017-02-22:
I am having a similar problem.  I have my "netxtion" folder in my library folder. (i.e. where the rest of the libraries are).  I really like this ease of use with this implementation and would like to use it.  My project is quite simple using 2 temperature probes, recipes and a servo motor to control a damper.

I am getting the following errors and wanted to know what in the H I am doing wrong.  I simply want to get this to compile.  I only tried to compile the CompPage.ino file.

/home/arnold/Arduino/libraries/Nextion/NexHardware.cpp:18:25: error: conflicting declaration ‘SoftwareSerial Serial2’
 SoftwareSerial nexSerial(10, 11); // RX, TX
                         ^
In file included from /usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:193:0,
                 from /home/arnold/Arduino/libraries/Nextion/NexHardware.h:17,
                 from /home/arnold/Arduino/libraries/Nextion/NexHardware.cpp:15:
/usr/share/arduino/hardware/arduino/cores/arduino/HardwareSerial.h:107:25: error: ‘Serial2’ has a previous declaration as ‘HardwareSerial Serial2’
   extern HardwareSerial Serial2;
                         ^
/home/arnold/Arduino/libraries/Nextion/NexHardware.cpp: In function ‘bool recvRetNumber(uint32_t*, uint32_t)’:
/home/arnold/Arduino/libraries/Nextion/NexHardware.cpp:70:31: warning: left shift count >= width of type [enabled by default]
         *number = (temp[4] << 24) | (temp[3] << 16) | (temp[2] << 8) | (temp[1]);
                               ^
/home/arnold/Arduino/libraries/Nextion/NexHardware.cpp:70:49: warning: left shift count >= width of type [enabled by default]
         *number = (temp[4] << 24) | (temp[3] << 16) | (temp[2] << 8) | (temp[1]);
                                                 ^

maxim25111, 2018-05-13:
You are trying to use define SoftwareSerial 2 tmes.
looks like nexSerial have already called SoftwareSerial as HardwareSerial included in file NexSerial.
They are conflicting together trying doing this same.
To understand why is error, open file nextHardware.h and NextHardware.cpp and search for HardwareSerial and Serial2. If there is included that names, it means that You dont need use it again in mine file. 
Or doing it arround, coment line where is Serial2 in that files HardwareSerial*

baugusto, 2018-11-20:
UNO-like Mainboards
If your board has only one hardware serial, such as UNO, you should disable dbSerial and redirect nexSerial to Serial(Refer to section:Serial configuration).

Open NexConfig.h and replace line 37 for this: #define nexSerial Serial
 and comment line 32, disabling dbSerial.

Then, recompile your code.

zhekaus, 2019-11-26:
That's weird to fix the library in order to make it work. I have different boards - Mega, UNO, etc.

alfonso2ff, 2023-02-04:
baugusto and JanitoRamos, thank you. After making corrections to Nexconfig.h the compilation ran without errors. Just a few minor warnings.

brunohermes, 2023-07-12:
> From this tuturial: https://randomnerdtutorials.com/nextion-display-with-arduino-getting-started/
> 
> This library is configured for Arduino MEGA2560 by default. To make it work for Arduino Uno, you need to do the following:
> 
> **1.** Open the ITEADLIB_Arduino_Nextion folder
> 
> **2.** There should be a _NexConfig.h_ file – open that file.
> 
> **3.** Comment line 27, so that it stays as follows:
> 
> //#define DEBUG_SERIAL_ENABLE **4.** Comment line 32:
> 
> //#define dbSerial Serial **5.** Change line 37, so that you have the following:
> 
> **#define nexSerial Serial** 6. Save the _NexConfig.h_ file.

THANK YOU SO MUCH!