itead/ITEADLIB_Arduino_Nextion #25: Examples are not working with NodeMCU v0.9

papaiatis, 2016-11-29:
I tried many of the examples with both an Arduino UNO and a NodeMCU v0.9 (ESP8266) and all of them worked in case of the UNO and none of them in case of the NodeMCU.
I connected the Nextion's RX to NodeMCU's D10 pin and Nextion's TX to NodeMCU's D9 pin. Also, I disabled Serial debugging and set `#define nexSerial Serial`
I found no relevant information on the Internet to solve this issue.

Any help would be appreciated!

sp9wun, 2017-05-10:
I see that nobody helped. I'm fighting with Nextion display and WeMos D1 R2 (ESP8266) since 3 days. I have 50% of success - I can send command to display and it's working very well. I connected it to D4 (TX for Serial1 in ESP8266 -> to RX in Nextion) and to D9 (RX for both serials in ESP8266 -> to TX in Nextion).
After long time of fighting with all sort of libraries I realised that I tried to use Latest ITEADLIB which won't work with ESP8266. After changing it to latest stable 0.7.0 version I reached small success. Exaple code "CompButton" now is compiling without errors and seems to work in one way. I see button and "1" digit on it. And in debug window I see:

```
RetCommandFinished err
recvRetCommandFinished err
setup done
```

It's long story - tried everything (softwareserial, external power supply (yes - I know about common GND), checked resistance of blue wire (Nextion TX)). Tried to send command: bkcmd=3 and page to see if something will came from Nextion. Without success. Also tried another libraries: https://github.com/bborncr/nextion. Finally, I bought USB2RS232 (CH340) converter and tomorrow I'll check display with Nextion Editor. But I suppose that this battle will be lost.

scropion86, 2017-08-23:
check this should solve your problem [](https://github.com/HobbytronicsPK/Nextion)