itead/ITEADLIB_Arduino_Nextion #30: NexUpload.cpp

Chris685, 2016-12-23:
It's not possible to compile for some Arduino Boards which are not compatible with the Libary "SoftwareSerial". Probably it would be good, to switch off the #include from the config-File.

File NexUpload.cpp.
17  //#include <SoftwareSerial.h>
18  
19  //#define USE_SOFTWARE_SERIAL 
20  #ifdef USE_SOFTWARE_SERIAL 

Comment out the Line 17 to compile for Arduino DUE which is using the 3 Uarts.

Thanks

DrStein99, 2017-07-13:
I had to remove the upload.* files out of my Nextion library directory, because my compiler was throwing out "SPI.H" errors, that I was too impatient to debug.  I never used this upload function  - and considering how reliable the rest of the library functions are, I am guessing this function works in the same unpredictable fashion.

vaewyn, 2019-01-22:
I just moved the #include into the #ifdef on the current version:
#ifdef USE_SOFTWARE_SERIAL
#include <SoftwareSerial.h>
SoftwareSerial dbSerial(3, 2); /* RX:D3, TX:D2 */
#define DEBUG_SERIAL_ENABLE
#endif

Works great with my ESP32 and 8266 then