itead/ITEADLIB_Arduino_Nextion #36: NexNumber setValue, getValue not working correctly

PaulChristopher, 2017-01-22:
MCU: Arduino ATMEGA 2560
Editor: V0.43
Library: V0.9.0

See attached test case: setValue and getValue do not work correctly for values bigger than 32767 on 8 bit systems.

To make at least setValue work correctly, I had to change NexNumber.cpp line 36 from 'utoa' to 'ultoa'. All in all, I am wondering if in setValue the size of the variable 'buffer' is big enough. The max value of uint32_t is supposed to be 4294967296. These are already 10 digits, and the string previx "obj.val=" is still missing. So 'buffer[10]' seems to be far too small and we are risiking a buffer overflow, aren't we?

With the above changes, I could make at least setValue work. But getValue still does not work although the code in NexHardware.cpp recvRetNumber, line 45 seems to be correct and ready for 4 byte uint32_t's. What's going on on the Nextion display side?

[testNextionNumberBox.zip](https://github.com/itead/ITEADLIB_Arduino_Nextion/files/721932/testNextionNumberBox.zip)

DrStein99, 2017-07-12:
I would also like to add, that the NUMBER class on the device, supports SIGNED integers.  Consider change to signed integer.

`bool NexNumber::setValue(int32_t number)`
`sprintf(buf, "%03i", number);`

aibnus, 2019-01-28:
To make getValue working properly I make a couple change:

1. In NexNumber.cpp change `bool NexNumber::getValue` to `uint32_t NexNumber::getValue`
2. In your code you have to declare number variable to `uint32_t`, and use this line `n0.getValue(&number)` instead `n0.getValue(number)`. 

example:
in global variable:
`uint32_t setSpeed;`

in main code:
`void b0PopCallback (void *ptr) {
  n0.getValue(&setSpeed);
}`

scls19fr, 2019-01-28:
Maybe you should consider sending a PR ;-)

lucianowindup, 2020-10-16:
I'm in trouble with getValue, it doesn´t work. I did what @aibnus commented, but no success for me. Is there any news about this matter?

lucianowindup, 2020-10-16:
`/*

Testes e verificação de componentes +Vida21:

- Blower e seu driver 12V
- Módulo controlador L298N
- Válvula solenóide 12V proporcional para O2
- Válvula solenoide 12V normalmente fechada para AR
- Display NEXTION 5 polegadas

Autor: Luciano Tanz
Data: 05/10/2020

*/

#include <Nextion.h>        //biblioteca Nextion

// Pinagem para o driver do blower
#define VSR_BLOWER      12  //PWM para controlar velocidade no VSR
#define EN_BLOWER       11  //HIGH/LOW para run/stop no EN (R min de 150 ohms p/ não queimar)
#define FR_BLOWER       10  //HIGH/LOW para rotação horária e anti-horária
#define FG_BLOWER        9  //Leitura de Pulsos em Hz no FG onde RPM=Pulsos*60/Polos, Polos=2
 
// Pinagem para o módulo L298N que aciona as válvulas   
#define IN1_VALVE_O2    8   //PWM para IN1 (IN2 em GND) p/ válvula proporcional O2 em OUT1 e OUT2
#define IN3_VALVE_AIR   7   //HIGH/LOW para IN3 (IN4 em GND) p/ abre/fecha válvula AR em OUT3 e OUT4

// Pinagem para sensores de pressão   
#define PRESSAO_1       A0  //entrada pressão 1
#define PRESSAO_2       A1  //entrada pressão 2

// Declaração objetos (Nextion Objeto var = Objeto(id página, id objeto, nome))

NexCheckbox c0 = NexCheckbox(0,8,"c0");   //checkbox para ligar blower
NexCheckbox c1 = NexCheckbox(0,9,"c1");   //checkbox para inverter blower
NexCheckbox c2 = NexCheckbox(0,10,"c2");  //checkbox para abrir ar

NexNumber n0 = NexNumber(0, 2,"n0");      //número 0..255 velocidade blower   
NexNumber n1 = NexNumber(0, 5,"n1");      //rotação rpm do blower
NexNumber n2 = NexNumber(0,12,"n2");      //número 0..255 válvula proporcional O2  
NexNumber n3 = NexNumber(0,14,"n3");      //pressão 1
NexNumber n4 = NexNumber(0,16,"n4");      //pressão 2  
NexNumber n5 = NexNumber(0,18,"n5");      //espelho para debug  

NexPage page0 = NexPage(0, 0, "page0");

// Variáveis globais

uint32_t val_vsr_blower;  //variavel PWM (0..255) da velocidade do blower 
uint32_t val_en_blower;   //variavel liga/desliga (HIGH/LOW) do blower 
uint32_t val_fr_blower;   //variavel rotação horária ou anti-horária do blower 
uint32_t val_fg_blower;   //variavel para rotação RPM do blower
 
uint32_t val_in1_valve_O2;   //variavel PWM (0..255) da valvula proporcional O2
uint32_t val_in3_valve_AIR;  //variavel abre/fecha (HIGH/LOW) da valvula de AR

uint32_t val_pressao_1;   //variavel pressão 1
uint32_t val_pressao_2;   //variavel pressão 2

long previoMillis = 0;
const int intervalo = 100;

// Inicialização
 
void setup() {

    Serial.begin(9600);
    //inicializa o TFT
    nexInit();
  
    //configura pinos para blower
    pinMode(VSR_BLOWER, OUTPUT);
    pinMode(EN_BLOWER, OUTPUT);
    pinMode(FR_BLOWER, OUTPUT);
    pinMode(FG_BLOWER, INPUT);

    //configura pinos para válvulas
    pinMode(IN1_VALVE_O2, OUTPUT);
    pinMode(IN3_VALVE_AIR, OUTPUT);

    //configura pinos para pressão
    pinMode(PRESSAO_1, INPUT);
    pinMode(PRESSAO_2, INPUT);

}

// Loop infinito

void loop()
{
  if(millis() - previoMillis >= intervalo)
  {
    previoMillis = millis();

    //Lendo valores para PWMs
    n0.getValue(&val_vsr_blower);
    n2.getValue(&val_in1_valve_O2);

    //Lendo checkboxes Nextion
    c0.getValue(&val_en_blower); 
      val_en_blower *= 255;
    c1.getValue(&val_fr_blower); 
      val_fr_blower *= 255;
    c2.getValue(&val_in3_valve_AIR); 
      val_in3_valve_AIR *= 255;
 
    //Controlando blower
    analogWrite(VSR_BLOWER, val_vsr_blower);
    digitalWrite(EN_BLOWER, val_en_blower);
    digitalWrite(FR_BLOWER, val_fr_blower);

    //Controlando válvulas
    analogWrite(IN1_VALVE_O2, val_in1_valve_O2);
    digitalWrite(IN3_VALVE_AIR, val_in3_valve_AIR);

    //Lendo e calculando rotação RPM do blower
    val_fg_blower = pulseIn(FG_BLOWER, HIGH);   //tempo do pulso em HIGH
    val_fg_blower += pulseIn(FG_BLOWER, LOW);   //mais o tempo de pulso em LOW
    val_fg_blower = 1000000*60/2/val_fg_blower;

    //Lendo pressões com map da faixa 0.5V (102) a 4.5V (921) para pressão min a max  
    val_pressao_1 = map(analogRead(PRESSAO_1),102,921,0,10);
    val_pressao_2 = map(analogRead(PRESSAO_1),102,921,0,20);
    
    //Atualizando RPM no Nextion
    n1.setValue(val_fg_blower);

    //Atualizando pressões no Nextion
    n3.setValue(val_pressao_1);
    n4.setValue(val_pressao_2);

    //Espelho para debug
    n5.setValue(val_vsr_blower+10);
  }    
}`