itead/ITEADLIB_Arduino_Nextion #38: utoa not included in standard C/C++ lib

hahnicity, 2017-03-18:
We're working on an Intel Edison which uses GNU libc/c++ and we've been trying to compile our nextion code but it doesn't work because `utoa` is not included in the base C/C++ lib. The only workaround I can currently think of is to modify the library to include `utoa`. Do you have any other suggestions?

DrStein99, 2017-07-13:
I modified my library to use itoa, since the number fields on the display will display signed integers - since I actually need to know negative or positive values on my readouts.  I suggest modifying the function to channel the data however it works for you.

Pharap, 2018-12-10:
Sorry to rake up such an old issue, but `utoa` (nonstandard) can be implemented with `sprintf``/`std::sprintf` (standard):
```cpp
char * utoa(unsigned int value, char * str, int radix)
{
	const char * format = nullptr;

	switch(radix)
	{
		case 8:
			format = "%o";
			break;
		case 10:
			format = "%u";
			break;
		case 16:
			format = "%x";
			break;
	}

	if(format == nullptr)
		return str;

	int size = std::sprintf(str, format, value);
	return &str[size];
}
```

Alternatively, if you only need radix 10 support and you can't afford `sprintf` then you can do:
```cpp
char * utoa(unsigned int value, char * str, int radix)
{
	if(radix == 10)
		while(value > 0)
		{
			unsigned int digit = value % 10;
			value /= 10;
			*str = static_cast<char>('0' + digit);
			++str;
		}

	return str;
}
```