itead/ITEADLIB_Arduino_Nextion #48: Comparison of alternatives libraries

scls19fr, 2018-03-05:
Hello,

I noticed that several libraries for Arduino and Nextion smart display exist.

- https://github.com/itead/ITEADLIB_Arduino_Nextion
- https://github.com/bborncr/nextion/
- https://github.com/DanNixon/NeoNextion

What are pros and cons of each libraries?

Pinging @shennongmin @bborncr @DanNixon

After enumerating differences, a paragraph about this could probably be added to https://github.com/itead/ITEADLIB_Arduino_Nextion/blob/master/readme.md

Kind regards

DanNixon, 2018-03-06:
When I first wrote my library the one provided by ITEAD was pretty poor in terms of functionality and general code quality. When I stopped maintaining it this was still the case.

https://github.com/bborncr/nextion/ looks rather incomplete in comparison to ITEAD's and my library, I can't really think of a use case for it (possibly low memory footprint, but at that point you are probably better of rolling your own library that only does what you need it to).

scls19fr, 2018-05-12:
License seems to be an important difference
- [itead/ITEADLIB_Arduino_Nextion](https://github.com/itead/ITEADLIB_Arduino_Nextion) is WTFPL v2 licensed (see https://github.com/itead/ITEADLIB_Arduino_Nextion/issues/49#issuecomment-372101704 about controversy :smile: )

https://github.com/itead/ITEADLIB_Arduino_Nextion/blob/5a5cf0abbf7f12e9bd98bb4540bbe8350bb5c9a3/html/index.html#L132-L134

- [DanNixon/NeoNextion](https://github.com/DanNixon/NeoNextion) is GPL v2 licensed 

https://github.com/DanNixon/NeoNextion/blob/7b0e6175bc20c7f7d8ee2424ca6bd3626666c3be/LICENSE#L1-L2

- [bborncr/nextion/](https://github.com/bborncr/nextion/) is GPL v3 licensed

https://github.com/bborncr/nextion/blob/bc4500ac54e9b52d937e268496b8e4500a3f51fd/LICENSE#L1-L2

intelligentandgoodlooking, 2018-05-14:
@DanNixon Thanks for making and sharing [NeoNextion](https://github.com/DanNixon/NeoNextion)
The original was (and probably still is) terrible. 
Especially needing to modify the library if you wanted to turn on debug messages or change the serial port. 
This obviously affected all projects that use the lib.
I started modifying the Nextion lib, then scrapped it when I found yours!
Nice contribution.

scls19fr, 2020-05-02:
https://github.com/VincentFrangi/Enhanced-Nextion-Library may be an interesting fork if the official library maintainers doesn't merge contributions
Maybe @VincentFrangi can tell us more about improvements he did

scls19fr, 2020-05-02:
https://github.com/jyberg/Enhanced-Nextion-Library from @jyberg

scls19fr, 2020-05-02:
I'm wondering if @DanNixon could dual licence 
https://github.com/DanNixon/NeoNextion under LGPL or BSD

DanNixon, 2020-05-02:
> I'm wondering if @DanNixon could dual licence
> [DanNixon/NeoNextion](https://github.com/DanNixon/NeoNextion) under LGPL or BSD

Why?

The Arduino framework is under GPL v2 so changing my library to something more permissive is going to benefit no one.

scls19fr, 2020-05-02:
I think the Arduino IDE is GPL but the Arduino Core is LGPL so it doesn't force user to use GPL for there own work (they can if the want but they can decide to use an other licence LGPL, BSD or even proprietary if they wish).
I like GPL software but prefer LGPL libraries.

scls19fr, 2020-05-02:
See https://www.arduino.cc/en/Main/FAQ
What do you mean by open-source hardware?

scls19fr, 2020-05-02:
See https://github.com/DanNixon/NeoNextion/issues/58