itead/ITEADLIB_Arduino_Nextion #51: where is library.properties file?

krzychoooo, 2018-05-04:
The eclipse environment can not see this library.

scls19fr, 2018-05-05:
itead should add this [library.properties](https://github.com/arduino/arduino/wiki/arduino-ide-1.5:-library-specification) file, tag a new release #47 , and ask at https://github.com/arduino/Arduino/issues?q=is%3Aissue+is%3Aopen+label%3A%22Component%3A+Board%2FLib+Manager%22 to add it to Arduino Library manager.

Maybe before this they should publish doc (#50 ) but my opinion is that they don't spend enough time working on development of this library ( #48 )

Even before, they should merge #26 ... for this... they just need to click on a green button named "Merge" ... and it's waiting since Dec 2016!

scls19fr, 2018-05-05:
Related issue #44

per1234, 2019-06-01:
Arduino Web Editor requires that all libraries (even those in the 1.0 library format such as this one) have a library.properties file. If you try to import a library without a library.properties file into the Arduino Web Editor, the import fails. The missing library.properties file, combined with the huge size of the library makes this very difficult for Arduino Web Editor users to use this library. This has already been reported a couple times on the Arduino Forum and I assisted those users with the relatively complex process of modifying the library so that it could be imported:
https://forum.arduino.cc/index.php?topic=618886

Here's the library.properties file I created:
```
name=ITEADLIB_Arduino_Nextion
version=0.7.0
author=Wu Pengfei
maintainer=ITEAD Intelligent Systems Co., Ltd.
sentence=An easy-to-use way to manipulate Nextion serial displays
paragraph=
category=Display
url=https://github.com/itead/ITEADLIB_Arduino_Nextion
architectures=*
includes=Nextion.h
```