itead/ITEADLIB_Arduino_Nextion #52: Ajust LIB ESP8266 With SDCard

laercionit, 2018-05-23:
/**
   @file NexUpload.cpp

   The implementation of download tft file for nextion.

   @author  Chen Zengpeng (email:<zengpeng.chen@itead.cc>)
   @date    2016/3/29
   @copyright
   Copyright (C) 2014-2015 ITEAD Intelligent Systems Co., Ltd. \n
   This program is free software; you can redistribute it and/or
   modify it under the terms of the GNU General Public License as
   published by the Free Software Foundation; either version 2 of
   the License, or (at your option) any later version.
*/

#include "NexUpload.h"
#include <SoftwareSerial.h>

//#define USE_SOFTWARE_SERIAL
#ifdef USE_SOFTWARE_SERIAL
SoftwareSerial dbSerial(3, 2); /* RX:D3, TX:D2 */
#define DEBUG_SERIAL_ENABLE
#endif

#ifdef DEBUG_SERIAL_ENABLE
#define dbSerialPrint(a)    dbSerial.print(a)
#define dbSerialPrintln(a)  dbSerial.println(a)
#define dbSerialBegin(a)    dbSerial.begin(a)
#else
#define dbSerialPrint(a)    do{}while(0)
#define dbSerialPrintln(a)  do{}while(0)
#define dbSerialBegin(a)    do{}while(0)
#endif

NexUpload::NexUpload(const char *file_name,uint32_t download_baudrate,const uint8_t SD_chip_select)
{
    _file_name = file_name; 
    _download_baudrate = download_baudrate;
    _SD_chip_select = SD_chip_select;
}

NexUpload::NexUpload(const String file_Name,uint32_t download_baudrate,const uint8_t SD_chip_select)
{
    NexUpload(file_Name.c_str(),download_baudrate,SD_chip_select);
}

NexUpload::NexUpload(const char *file_name,uint32_t download_baudrate)
{
    _file_name = file_name; 
    _download_baudrate = download_baudrate;
    _SD_chip_select = NULL;
}

NexUpload::NexUpload(const String file_Name,uint32_t download_baudrate)
{
    NexUpload(file_Name.c_str(),download_baudrate);
}

void NexUpload::upload(void)
{
  dbSerialBegin(115200);
  if (!_checkFile())
  {
    dbSerialPrintln("the file is error");
    return;
  }
  if (_getBaudrate() == 0)
  {
    dbSerialPrintln("get baudrate error");
    return;
[... 311 lines cut ...]
    /*
     * Receive string data. 
     * 
     * @param buffer - save string data.  
     * @param timeout - set timeout time. 
     * @param recv_flag - if recv_flag is true,will braak when receive 0x05.
     *
     * @return the length of string buffer.
     *
     */   
    uint16_t recvRetString(String &string, uint32_t timeout = 100,bool recv_flag = false);
    
private: /* data */ 
    uint32_t _baudrate; /*nextion serail baudrate*/
    const char *_file_name; /*nextion tft file name*/
    File _myFile; /*nextion ftf file*/
    uint32_t _undownloadByte; /*undownload byte of tft file*/
    uint8_t _SD_chip_select; /*sd chip select pin*/
    uint32_t _download_baudrate; /*download baudrate*/
};
/**
 * @}
 */

#endif /* #ifndef __NEXDOWNLOAD_H__ */

rcpehls, 2019-06-06:
Hello @laercionit 

I used your ESPNextionUploadHDSerialSD to upload a TFT file located on my SPIFFS but connecting my IHM and starting my esp8266 I received on monitor the message below.

ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld 

#### my code

`#include "NexUpload.h"

NexUpload nex_up("test_ihm.tft",115200);

void setup() {
    nex_up.upload();
}

void loop() {}`

#### my device

Device: Wemos D1 mini PRO
CPU Freq: 80mhz
Flash: 16mb

laercionit, 2019-06-06:
Hello, a library I put to use TFT of the SD Card, is not prepared to use the SPIFFS.

rcpehls, 2019-06-06:
> Hello, a library I put to use TFT of the SD Card, is not prepared to use the SPIFFS.

Ok I will add the feature so that it can be uploaded by SPIFFS.

Thanks for the reply.