itead/ITEADLIB_Arduino_Nextion #56: getValue doesnt work

m50008, 2018-10-12:
When using the getvalue() method to get a value from a slider, i get the serial output.
get h0.val���recvRetNumber err
source code used
uint32_t number = 0;
char temp[10] = {0}; 
h0.getValue(&number);
utoa(number, temp, 10);

llist, 2018-11-06:
Do you have a Nextion screen ?
how many screens did you draw ?
When you try to get the value of h0, the screen that contains the variable h0 is it in foreground  ?

jyberg, 2019-02-26:
Hi, 
Is you slide global and have you other page open when you request the value. This version of library not support global objects, and has many other issues also, as this library is not well maintained and pull request are not handeld here link to Enhanced library version:
https://github.com/jyberg/Enhanced-Nextion-Library There is also other project that setups NodeMcu/Esp8266 wifi settings via Nextion UI for reference...

daseri, 2019-03-19:
Hi jyberg. 
I´m trying your libraries and have an issue compiling for arduino uno 
I have errors like this:
NexWaveform.h:20:18: fatal error: vector: No such file or directory
 #include <vector>

what is the problem?

jyberg, 2019-03-19:
Hi, it looks like you don't have std::vector include file in yous system, or vector include file is with different name or so.
first try to comment that line away if it then gives error in line where std::vector<T> is used then std:vector definitios is not available. Above link has similar issue.
https://arduino.stackexchange.com/questions/24247/current-arduino-stl-libraries
There was said that you need add #include "StandardCplusplus.h" before #include <vector> line
one thing is also if you don't need above function you can comment #include <vector> and above function away.
    template<typename T>
bool addValues(uint8_t ch, std::vector<T> &values)
{
...
}

daseri, 2019-03-19:
Hi again.
Thanks for your reply. 
I added the standarC++ library and now I`ve a lot of warnings like this:
/NexHardware.h:34:13: error: 'function' in namespace 'std' does not name a template type
 extern std::function<void(uint8_t)> currentPageIdCallback;

jyberg, 2019-03-19:
No it's not std::function is defined in functional header file #inluclude <functional>
This was my fault I had #include <functioanal> in NextHardware.cpp file move it from there to NextHardware.h file after #include <arduino>, In my projects functional is automatically included in some other system include files. That should help. std::function template class has been in C++ since C++ 11 standard

daseri, 2019-03-20:
Hi!

I've the same warnings. 
I'd like to try your libraries but it´s imposible to me. Could you send to me the libraries you`ve installed in your IDE?  vector, functional .. etc?
Regards!

jyberg, 2019-03-20:
Hi,
I have done all with Platform IO and for ESP8266 board... I just made Arduino UNO project and now I see your problems, it seems and ESP8266 framework is much rich than base arduino, that misses most of base std classes, that makes life more easy... I will to changes that will remove std need... be tuned..

daseri, 2019-03-20:
Oh. Thank you very much. I Stay tuned to updates!
;)
Regards!

jyberg, 2019-03-20:
Hi, Now there is updated version available c++ std:: usage can be enablen/disabled from NexConfig.h
I also added C style function variant for functions that used std::vector<> + I chagned std::function pointers declarations to C style function pointers.

I did only very base Arduino project where I tried compilation, if any issues let me know...

daseri, 2019-03-20:
Great!

I can compile now!
I´ll check all examples according to know how to work with setValue and getValue in not shown pages. Regards!
;)

jyberg, 2019-03-20:
Global variables you need to set page parameter like: NexText gSsid(4,39,"ssid", &pWifi);
Where "&pWifi" is pointer to NexPage pWifi(4,0,"wifi"); page definition. Page parameter is optional and then variable can be used as local only. In Nextion Editor you must define variables as gloal if you want set value when page is not active.

daseri, 2019-03-20:
Thanks for the info!!
Now I having problems with this:
warning: invalid conversion from 'uint32_t* {aka long unsigned int*}' to 'int32_t* {aka long int*}' [-fpermissive]

   on_hour.getValue(&configON_hour);

Did you change the getValue and SetValue type of data?
I always get value "0" and in serial monitor I get messages like "????????" when getting and setting values 
In the oficial library this issue didn´t appear
Do you know how can be the problem?

jyberg, 2019-03-20:
Yes I corrected functions. Value is not uint32 in nextion it is int32 negative values supported and get functions return value is bool to indicate success/failure. Check function definition in .h file there are in/out and return values defined.

daseri, 2019-03-22:
Hi again jyberg!

I,ve my nextion working with your libraries! I can read and write values in any page! Is great!
Now I`m having troubles with rtc. If I try with the example I´ve this warnings. 
 warning: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*' [-fpermissive]

     rtc.read_rtc_time(time_buf,30);

Have you any idea? 
I,ve working correctly rtc with the oficial libraries.

jyberg, 2019-03-23:
Hi,
I have not used rtc functions but now when cheking whole rtc implementation is quite poort.. But becaue I don't have enhanced mode, so re-implementing without testing is not good idea. but your problem is that in example time_buf data type is uint8_t, and function requires char data type, so change uint8_t time_buf[30] = {0}; to char time_buf[30]={0}; Because done rtc interface / functions is quite exotic, you must read from NexRtc.cpp file how the functions works...

daseri, 2019-03-23:
Yes I checked it after answer, sorry, was my fault.
I,d working well now if I only work with the RTC code but seems to be a problem with serial port. 
My main code works properly, but when I use rtc.read_rtc_time(time_buf, 30); funcion in any part of my code, serial port is always reading (tx light turned on all time) and I the getValues or setValues in serial monitor as "?????????????" and nothing works properly. 
The read_rtc_time function is cracking all even if the function is in a part of code that don´t execute 
It´s crazy

jyberg, 2019-03-23:
try to do rtc variable reads by your self like:
    cmd = "get rtc0";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&year))
    {
        return false;
    }
or add your own debug info in the nextion interface code so that you can see what happens, for me these rtc variables looks like any other variable in the system, only thing is that they are only available in enhanced necxtion models

edurmaciel, 2019-03-31:
> Hi again jyberg!
>
> I,ve my nextion working with your libraries! I can read and write values in any page! Is great!
> Now I`m having troubles with rtc. If I try with the example I´ve this warnings. [...]

Hello @jyberg !

I am having the same error with getValue returning err.
I checked your pull request(which apparently was not approved yet, or am I wrong here?) trying to identify your changes but I am not that experienced.

Would you mind sharing the fix? The whole library, in the case :) 

Thank you very much in advance.

edurmaciel, 2019-03-31:
```
#include "Nextion.h"

NexNumber n9 = NexNumber(1, 3, "n9"); 

uint32_t number = 0; 
  
void setup() {
  nexInit();
}

void loop() {
  
  n9.getValue(&number);
  Serial.println(number,DEC);
  
  //if(number==0)
  //{
  //n7.setValue(2);
  //}
}
```

The code above produces the following output:

```0
get n9.val⸮⸮⸮recvRetNumber err
0
get n9.val⸮⸮⸮recvRetNumber err
0
```

Just so you can understand what I am doing, in the Nextion side, I have a timer snippet, constantly changing the objects (n7, n8 and n9) values. It is a clock which I want the arduino to periodically check and set/reset :

```
n9.val++
if(n9.val==60)
{
 n9.val=0
 n8.val++
 if(n8.val==60)
 {
   n8.val=0
   n7.val++
   if(n7.val==24)
   {
     n8.val=0
   }
 }
}
```

jyberg, 2019-04-01:
Hi, are you using pull request version, or version directly from my git page. As you can see this version is not maintained, and I don't belive, that pull request will be accepted, because lack of interest. So I recommend to use version from my git page, because I have correcte multiple issues after pull request. When checking your code try to make following change NexNumber n9 = NexNumber(1, 3, "n9");  change to NexNumber n9(1,3,"n9"); because Nex.... classes has no assigment operator/copy constructor defined.

edurmaciel, 2019-04-01:
Thank you for your reply @jyberg !
I will try with the version in your git later today. This one: https://github.com/jyberg/Enhanced-Nextion-Library
Fingers crossed :)

edurmaciel, 2019-04-01:
@jyberg is your library still supporting arduino leonardo or our changes are specific to esp8622/nodeMCU ?

jyberg, 2019-04-02:
yes there is nothing board specific, only thing is that don't enable STL library usage if you don't have STL library implementation available. (NodeMCU includes STL library implementation but in base Arduino it is not included by default)

14dieseloutlaw14, 2019-08-10:
> yes there is nothing board specific, only thing is that don't enable STL library usage if you don't have STL library implementation available. (NodeMCU includes STL library implementation but in base Arduino it is not included by default)

Im trying to use your version of the libraries with my Mega. 
changed this 
```
//#define NEX_SOFTWARE_SERIAL
#ifndef NEX_SOFTWARE_SERIAL
// hardware Serial port
#define nexSerial Serial2
```

to use tx and rx 2 ports 

but it wont compile any suggestions ?

14dieseloutlaw14, 2019-08-10:
this is the error

```
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\Print.cpp: In function 'print.constprop':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\Print.cpp:75:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "doxygen.h"
 Used: C:\Users\chive\OneDrive\Documents\Arduino\libraries\Enhanced-Nextion-Library-master
 Not used: C:\Users\chive\OneDrive\Documents\Arduino\libraries\ITEADLIB_Arduino_Nextion-master
 Not used: C:\Users\chive\OneDrive\Documents\Arduino\libraries\ITEADLIB_Arduino_Nextion-0.7.0
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

jyberg, 2019-08-12:
Hi,

I can't repeat the problem I used same Windows Store Arduino ide and same board option. I build enhanced library \examples\CompText example without problems. And yes I changed nexSerial pointin to Serial2

br,
Jyrki

14dieseloutlaw14, 2019-08-14:
i was able to get it to compile.. but get value still doesnt work for me,

```
void b14PushCallback (void *ptr) // Press event for ride height setting 1
{
  uint32_t setval;
  
  n40.getValue(&setval);
 
  Serial.print(setval);
}
```

14dieseloutlaw14, 2019-08-14:
here is the error

```
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1200:3,14,(null)b13]
[1186:3,15,(null)b14]
recvRetNumber err
200[1186:3,15,(null)b14]
recvRetNumber err
200
```

14dieseloutlaw14, 2019-08-14:
now im getting this??

```
[1308:3,15,(null)b14]
recvRetNumber err
recv command err value: 26
recvRetCommandFinished err
53659659[1518:3,0,(null)page3]
[1308:3,15,(null)b14]
recvRetNumber err
recv command err value: 26
recvRetCommandFinished err
53659659
```

jyberg, 2019-08-14:
hi,
Main source for Nextion commands, and return values is https://nextion.tech/instruction-set/
From there error code dec: 26 -> hex: 0x1A

Invalid Variable name or attribute: Returned when invalid Variable name or invalid attribute was used

br,
Jyrki

14dieseloutlaw14, 2019-08-14:
Is this because the b14 is on a different page than the n40 box?

jyberg, 2019-08-14:
Hi,

I don't know your project. In general you can call only objects in active page. Exeption on that is if object is global. But the  you must define page information in library object in your code object initialization.

14dieseloutlaw14, 2019-08-14:
Ok so i have to figure out a way to bring up the page 3 that has n40 on it, when i press b14 on page 4 or have invisible number boxes on page 4 with the button im pressing. I have a total of 12 of these number boxes im trying to apply this to

14dieseloutlaw14, 2019-08-15:
ok so i was able to get the values while i was on the page that they were on but if possible can you explain what you ment by  this quote?

jyberg, 2019-08-15:
so,
if you define object in nextion editor as global (vscope parameter) then object value can be set/read from any page, by defining page in set/get request.
In Library I added this support in the object classess
 NexObject::NexObject(uint8_t pid, uint8_t cid, const char *name, const NexObject* page=nullptr);
where NexObject* page parameter is pointer to the page where object is located in UI.

e.g.
// page 4 wifi
NexPage pWifi(4,0,"wifi");
//two global text object in the page wifi
NexText gSsid(4,39,"ssid", &pWifi);
NexText gPwd(4,41,"pwd", &pWifi);

Now when gSsid and gPwd objects has information about page library sends commands including optional page information to the nextion UI.

14dieseloutlaw14, 2019-08-15:
Thank you so much... this has sovled mu issues

lucianowindup, 2020-10-17:
I have had problems with getValue. It works with Nextion Editor (debug with user MCU Input), but doesn´t work with the real display. I tried the ordinary library of ITEADLIB and your LIB @jyberg , but no success. Could you help me?
`
#include "Nextion.h"

// Pinagem para o driver do blower
#define VSR_BLOWER      12  //PWM para controlar velocidade no VSR
#define EN_BLOWER       11  //HIGH/LOW para run/stop no EN (R min de 150 ohms p/ não queimar)
#define FR_BLOWER       10  //HIGH/LOW para rotação horária e anti-horária
#define FG_BLOWER        9  //Leitura de Pulsos em Hz no FG onde RPM=Pulsos*60/Polos, Polos=2
 
// Pinagem para o módulo L298N que aciona as válvulas   
#define IN1_VALVE_O2    8   //PWM para IN1 (IN2 em GND) p/ válvula proporcional O2 em OUT1 e OUT2
#define IN3_VALVE_AIR   7   //HIGH/LOW para IN3 (IN4 em GND) p/ abre/fecha válvula AR em OUT3 e OUT4

// Pinagem para sensores de pressão   
#define PRESSAO_1       A0  //entrada pressão 1
#define PRESSAO_2       A1  //entrada pressão 2

// Declaração objetos (Nextion Objeto var = Objeto(id página, id objeto, nome))

// Declare Nextion instance
Nextion *next = Nextion::GetInstance(Serial); // HW serial

NexCheckbox c0 = NexCheckbox(next,0,8,"c0");   //checkbox para ligar blower
NexCheckbox c1 = NexCheckbox(next,0,9,"c1");   //checkbox para inverter blower
NexCheckbox c2 = NexCheckbox(next,0,10,"c2");  //checkbox para abrir ar

NexNumber n0 = NexNumber(next,0, 2,"n0");      //número 0..255 velocidade blower   
NexNumber n1 = NexNumber(next,0, 5,"n1");      //rotação rpm do blower
NexNumber n2 = NexNumber(next,0,12,"n2");      //número 0..255 válvula proporcional O2  
NexNumber n3 = NexNumber(next,0,14,"n3");      //pressão 1
NexNumber n4 = NexNumber(next,0,16,"n4");      //pressão 2  
NexNumber n5 = NexNumber(next,0,18,"n5");      //espelho para debug  

NexPage page0 = NexPage(next, 0, "page0");

// Variáveis globais

uint32_t val_vsr_blower;  //variavel PWM (0..255) da velocidade do blower 
uint32_t val_en_blower;   //variavel liga/desliga (HIGH/LOW) do blower 
uint32_t val_fr_blower;   //variavel rotação horária ou anti-horária do blower 
uint32_t val_fg_blower;   //variavel para rotação RPM do blower
 
uint32_t val_in1_valve_O2;   //variavel PWM (0..255) da valvula proporcional O2
uint32_t val_in3_valve_AIR;  //variavel abre/fecha (HIGH/LOW) da valvula de AR

uint32_t val_pressao_1;   //variavel pressão 1
uint32_t val_pressao_2;   //variavel pressão 2

long previoMillis = 0;
const int intervalo = 100;

// Inicialização
 
void setup() {

    Serial.begin(9600);
    //inicializa o TFT
    next->nexInit(9600);
  
    //configura pinos para blower
    pinMode(VSR_BLOWER, OUTPUT);
    pinMode(EN_BLOWER, OUTPUT);
    pinMode(FR_BLOWER, OUTPUT);
    pinMode(FG_BLOWER, INPUT);

    //configura pinos para válvulas
    pinMode(IN1_VALVE_O2, OUTPUT);
    pinMode(IN3_VALVE_AIR, OUTPUT);

    //configura pinos para pressão
    pinMode(PRESSAO_1, INPUT);
    pinMode(PRESSAO_2, INPUT);

}

// Loop infinito

void loop()
{
  if(millis() - previoMillis >= intervalo)
  {
    previoMillis = millis();

    //Lendo valores para PWMs
    n0.getValue(&val_vsr_blower);
    n2.getValue(&val_in1_valve_O2);

    //Lendo checkboxes Nextion
    c0.getValue(&val_en_blower); 
      val_en_blower *= 255;
    c1.getValue(&val_fr_blower); 
      val_fr_blower *= 255;
    c2.getValue(&val_in3_valve_AIR); 
      val_in3_valve_AIR *= 255;
 
    //Controlando blower
    analogWrite(VSR_BLOWER, val_vsr_blower);
    digitalWrite(EN_BLOWER, val_en_blower);
    digitalWrite(FR_BLOWER, val_fr_blower);

    //Controlando válvulas
    analogWrite(IN1_VALVE_O2, val_in1_valve_O2);
    digitalWrite(IN3_VALVE_AIR, val_in3_valve_AIR);

    //Lendo e calculando rotação RPM do blower
    val_fg_blower = pulseIn(FG_BLOWER, HIGH);   //tempo do pulso em HIGH
    val_fg_blower += pulseIn(FG_BLOWER, LOW);   //mais o tempo de pulso em LOW
    val_fg_blower = 1000000*60/2/val_fg_blower;

    //Lendo pressões com map da faixa 0.5V (102) a 4.5V (921) para pressão min a max  
    val_pressao_1 = map(analogRead(PRESSAO_1),102,921,0,10);
    val_pressao_2 = map(analogRead(PRESSAO_1),102,921,0,20);
    
    //Atualizando RPM no Nextion
    n1.setValue(val_fg_blower);

    //Atualizando pressões no Nextion
    n3.setValue(val_pressao_1);
    n4.setValue(val_pressao_2);

    //Espelho para debug
    n5.setValue(val_en_blower+10);
  }    
}
`