itead/ITEADLIB_Arduino_Nextion #59: getText does not work on text box

RishabhHardas, 2018-11-29:
I was working on interfacing a Nextion Display with Arduino Mega.

The GUI has 3 Pages. The first page is used to display data, while the other two are used to take inputs and set some parameters.

After the user inputs data into the text box, I tried to get the value of the text box with the getText() method which is a member function of the NexText calss.

The method works fine if the text box is on the first page, but fails if it is on the second or the third page.

When getting the values the screen in not in the foreground on the Nextion Display

kilicaslanyunus, 2018-12-27:
Its same for me. Could you solve the problem?

RishabhHardas, 2018-12-27:
The only workaround I found was to have the page in the foreground and fetch the value.
You can use the sendCommand() function to bring the page to the foreground, fetch the value and go back to the page you were at.

jyberg, 2019-02-26:
Hi, as you can see this project is not well managed. I have done pull request that corrects this and many other problems in the library, but I don't belive that reques will be accepted soon.... The problem is that you are using global variables and this libary version not support it. Library not include page name in value request. My Enhanced Nextion library will fix this and many other issues... https://github.com/jyberg/Enhanced-Nextion-Library I hope this helps, it was takin some time to solve out all of these problems that I fixed... ;) There is also example NodeMcu Nextion wifi setup project if you want learn more...

traxxtar, 2019-03-12:
I added your library and it won't compile at all. The below is just a sample of the errors, much more are in the compile output screen.

`Users/Darren/Documents/Hobbies/Arduino/LaserMeasure/Laser_v0.10/Laser_v0.10.ino: In function 'void btnenterPushCallback(void*)':
Laser_v0.10:118:31: error: invalid initialization of non-const reference of type 'uint16_t& {aka unsigned int&}' from an rvalue of type 'unsigned int'
   temp1.getText(buffer, sizeof(buffer));  // get distance data the user entered
                               ^
In file included from /Users/darren/Documents/Arduino/libraries/DLM-Nextion-Library/dNextion.h:32:0,
                 from /Users/Darren/Documents/Hobbies/Arduino/LaserMeasure/Laser_v0.10/Laser_v0.10.ino:33:
/Users/darren/Documents/Arduino/libraries/DLM-Nextion-Library/NexText.h:60:10: note:   initializing argument 2 of 'bool NexText::getText(char*, uint16_t&)'
     bool getText(char *buffer, uint16_t &len);

          ^
/Users/Darren/Documents/Hobbies/Arduino/LaserMeasure/Laser_v0.10/Laser_v0.10.ino: In function 'void btnLJmeasurePushCallback(void*)':
Laser_v0.10:168:38: error: invalid initialization of non-const reference of type 'uint16_t& {aka unsigned int&}' from an rvalue of type 'unsigned int'
     txtoffset1.getText(butter, sizeof(butter));  // get offset data
                                      ^
In file included from /Users/darren/Documents/Arduino/libraries/DLM-Nextion-Library/dNextion.h:32:0,
                 from /Users/Darren/Documents/Hobbies/Arduino/LaserMeasure/Laser_v0.10/Laser_v0.10.ino:33:
/Users/darren/Documents/Arduino/libraries/DLM-Nextion-Library/NexText.h:60:10: note:   initializing argument 2 of 'bool NexText::getText(char*, uint16_t&)'
     bool getText(char *buffer, uint16_t &len);`

jyberg, 2019-03-16:
getText len parameter is reference type variable, so you can't use sizeof(... because it can't be used as reference. When getText returns len parameter includes received string length exluding null character.
So do following chagne:
uint16_t len=sizeof(buffer);
getText(buffer, len);

Orthodudz, 2019-05-23:
> getText len parameter is reference type variable, so you can't use sizeof(... because it can't be used as reference. When getText returns len parameter includes received string length exluding null character.
> So do following chagne:
> uint16_t len=sizeof(buffer);
> getText(buffer, len);
> 
> I did this type of correction to the library quite mutch so that get functions return bool for success/fail support and function reference parameter are used for in/out information for function calls. e.g this getText. it returned string length earlier. There was no information is call successfully completed (OK checking return value >0 but what if there is no text in componet? then txt len is 0!). So return value is bool, len parameter tells in case of success leght of returned text excluding null char.

Hi jyberg,

I am a totally newbie in coding. I deleted the old library of Nextion and used your modified library. However I still cannot wonder why I cannot get data using getText() method. See attached
[Output counter with Nextion24.txt](https://github.com/itead/ITEADLIB_Arduino_Nextion/files/3210785/Output.counter.with.Nextion24.txt)
 code. Please notify me what I am missing.
Thanks.

jyberg, 2019-05-23:
Hi, did you take latest version from my github page?
P0SensorSV.getText(... returns true if success so if reading success from nextion then return value os true. also you can't compare char arrays like this "if(txtSensorSV!=txSensorSV)", and I don't understand what you wan't achieve with this  "txSensorSV = txtSensorSV;"
have you checked basics e.g. from https://www.learn-c.org/ or https://www.tutorialspoint.com/cprogramming/c_basic_syntax.htm

Orthodudz, 2019-05-24:
> Hi, as you can see this project is not well managed. I have done pull request that corrects this and many other problems in the library, but I don't belive that reques will be accepted soon.... The [...]

Hi jyberg, I downloaded yesterday the version from your github that was on your reply here Feb. 27. :) I am sorry but I really do not know how to compare char array. I thought I can get a result with this "if(txtSensorSV!=txSensorSV)", and if the result is true I save the value from "txtSensorSV" to "txSensorSV" with this "txSensorSV = txtSensorSV;". Then to see that getText is working, I print a word "Different" to a text component on page1 with this "P1IPAdd.setText("Different");"

jyberg, 2019-05-24:
that why I pointed some C instructions so see https://www.learn-c.org/en/Strings
or see more and learn form C / C++ reference pages...
http://pubs.opengroup.org/onlinepubs/9699919799/nframe.html
http://www.cplusplus.com/reference/cstring/strcmp/?kw=strcmp
https://en.cppreference.com/w/cpp/string/byte/strcmp

Orthodudz, 2019-05-24:
Thank you jyberg. I try to learn from these links.

Orthodudz, 2019-05-28:
> that why I pointed some C instructions so see https://www.learn-c.org/en/Strings
> or see more and learn form C / C++ reference pages...
> http://pubs.opengroup.org/onlinepubs/9699919799/nframe.html [...]

Hi jyberg! Thanks alot. I finally make my test work. But I used the previous library because your library does not allow me to get text value with more than 2 characters. Can you check it please what is missing in my code if i use your library?
 I attached here my arduino code and the result pictures using your library and the previous library for your reference. 
[Output counter with Nextion24.txt](https://github.com/itead/ITEADLIB_Arduino_Nextion/files/3225445/Output.counter.with.Nextion24.txt)
[image]
[image]

jyberg, 2019-05-28:
P0SensorPV.setText(itoa(analogRead(reflectiveSense), txtSensorPV, 10));
  txtSensorSV[0]=0;
  P0SensorSV.getText(txtSensorSV, LentxtSensorSV);
    if(strncmp(txtSensorSV, txSensorSV, sizeof (txtSensorSV)) !=0)
    {
      strncpy(txSensorSV, txtSensorSV, sizeof (txSensorSV));
      P1Status.setText(txSensorSV);
      P1IPAdd.setText("Different");
    }
    else{P1IPAdd.setText("Same");}

sanh10, 2019-07-11:
Hi.. plz can any1 tell me how to read the text from nextion enhanced model using Arduino & display on arduino serial monitor

jyberg, 2019-07-15:
Hi,

are you using enhanced version from here itead github page, or do you use version from my github repository? I recommend to use my repository version because it has more fixes than in here my merge request version. Itead not maintain this version. I also hope that future questions/issue to enhanced version gomes via my repo.

Then about reading and setting the values. There is quite good examples in https://github.com/jyberg/Esp8266-Nextion-Wifi-Setup project. see: esp8266\src\nextionUi.cpp

sanh10, 2019-07-15:
Ok I'll go through this ..thank you

sanh10, 2019-07-20:
Hi sir,
Can you tell me how to convert string value & display on nextion display. I
am struck at this point & unable to display the value on nextion display.

Regards,
Santosh s hiremath,
Email: santoshhiremath88@gmail.com

sanh10, 2019-07-20:
Hi sir,
Can you tell me how to convert string value & display on nextion display. I am struck at this point & unable to display the value on nextion display.

jyberg, 2019-07-23:
Do you mean this how to get char* from String object. use String object c_str() method

NexText nexTxt(0,1,"nexTxt");
String msg;
msg = "ABBA";
nexTxt.setText(msg.c_str())

messplay, 2022-08-03:
Well I modified the nextion library and obtain string of the text box

in nextText.h add
`String getText2(char *buffer, uint16_t len);`

in nextext.cpp add
`String NexText::getText2(char *buffer, uint16_t len)
{
    String cmd;
    cmd += "get ";
    cmd += getObjName();
    cmd += ".txt";
    sendCommand(cmd.c_str());
    return recvRetString2(buffer,len);
}`

in nextHardware.h add
`String recvRetString2(char *buffer, uint16_t len, uint32_t timeout = 100);`

and finally in nextHardware.cpp
`String recvRetString2(char *buffer, uint16_t len, uint32_t timeout)
{
    uint16_t ret = 0;
    bool str_start_flag = false;
    uint8_t cnt_0xff = 0;
    String temp = String("");
    uint8_t c = 0;
    long start;

    if (!buffer || len == 0)
    {
        goto __return;
    }
    
    start = millis();
    while (millis() - start <= timeout)
    {
        while (nexSerial.available())
        {
            c = nexSerial.read();
            if (str_start_flag)
            {
                if (0xFF == c)
                {
                    cnt_0xff++;                    
                    if (cnt_0xff >= 3)
                    {
                        break;
                    }
                }
                else
                {
                    temp += (char)c;
                }
            }
            else if (NEX_RET_STRING_HEAD == c)
            {
                str_start_flag = true;
            }
        }
        
        if (cnt_0xff >= 3)
        {
            break;
        }
    }

    ret = temp.length();
    ret = ret > len ? len : ret;
    strncpy(buffer, temp.c_str(), ret);
    
__return:

    dbSerialPrint("recvRetString[");
    dbSerialPrint(temp.length());
    dbSerialPrint(",");
    dbSerialPrint(temp);
    dbSerialPrintln("]");

    return buffer;
}`