itead/ITEADLIB_Arduino_Nextion #68: ESP32 / Nextion connection error

pfeno, 2019-02-25:
Hi,
I have some problem to communicate between my ESP32 and my nextion screen.
I try with a default example _CompButton_v0_32_ and when I call `nexinit()` I have a false error code `recvRetCommandFinished err`

I investigate and modify NexHardware.cpp  to show the result from the display.
In function `bool recvRetCommandFinished(uint32_t timeout)`, I just add 1 line :

>         for (int i = 0; i < 4; i++) Serial.println(temp[i], HEX);

The result, in the Arduino serial monitor was very strange for me and corresponding to nothing:
`recvRetCommandFinished err
FE
FF
FF
62

0
recvRetCommandFinished err
70
61
67
65

0`

Could you help me ?
Regards
nomi.fenoglio@arcelormittal.com

PS : I work on an EP32 (v1.0.1), ITEADLIB Arduino Nextion library get from github (v0.9.0)

jyberg, 2019-02-25:
I had also lot of problems with this library version. It is no well maintained, and examples not work properly.
You may wan't to check my fork and NodeMcu/Esp8266 Wifi setup via Nextion UI as reference and starting point: https://github.com/jyberg/Esp8266-Nextion-Wifi-Setup  or only enhancend Nextion library version: https://github.com/jyberg/ITEADLIB_Arduino_Nextion

br,
Jyke

pfeno, 2019-02-26:
@jyberg I think I have also some issue with your library.

First, it's only for ESP8266 ! Think not very difficult to adapt for ESP32.
Secondly, you continue to use the default Nextion lib, and for me, just with the `nexInit()` API, I have error as you can read on my first post.

br,
Nomi

jyberg, 2019-02-26:
hi, see https://www.itead.cc/wiki/Nextion_Instruction_Set for Nextion commands and retrun values,
for me it looks like the problem is not the library, instead you receive some unnoud data, so check Serial communication so that no other components thatn Nextion can send data to RX connection.  First post 70 61 67 65 hex code is in text "page" and later 72 65 63 is "rec"

pfeno, 2019-02-27:
> hi, see https://www.itead.cc/wiki/Nextion_Instruction_Set for Nextion commands and retrun values,
> for me it looks like the problem is not the library, instead you receive some unnoud data, so check [...]

Yes, I saw that just before leaving works!
I check today, and reinstall the library if necessary.
Thanks for your support.
br

pfeno, 2019-02-27:
@jyberg I reinstall the library .. and same penalty.
I try another example and idem.
But, strangely, the callback function work.

```
/**
 * @example CompButton.ino
 * 
 * @par How to Use
 * This example shows that when the button component on the Nextion screen is released,
 * the text of this button will plus one every time.      
 *
 * @author  Wu Pengfei (email:<pengfei.wu@itead.cc>)
 * @date    2015/7/10
 * @updated 2016/12/25 bring HMI up to v0.32 to avoid too old issues
 * @convert by Patrick Martin, no other changes made
 * @copyright 
 * Copyright (C) 2014-2015 ITEAD Intelligent Systems Co., Ltd. \n
 * This program is free software; you can redistribute it and/or
 * modify it under the terms of the GNU General Public License as
 * published by the Free Software Foundation; either version 2 of
 * the License, or (at your option) any later version.
 */

#include "Nextion.h"

/*
 * Declare a button object [page id:0,component id:1, component name: "b0"]. 
 */
NexButton b0 = NexButton(0, 1, "b0");

char buffer[100] = {0};

/*
 * Register a button object to the touch event list.  
 */
NexTouch *nex_listen_list[] = 
{
    &b0,
    NULL
};

/*
 * Button component pop callback function. 
 * In this example,the button's text value will plus one every time when it is released. 
 */
void b0PopCallback(void *ptr)
{
    uint16_t len;
    uint16_t number;
    NexButton *btn = (NexButton *)ptr;
    dbSerialPrintln("b0PopCallback");
    dbSerialPrint("ptr=");
    dbSerialPrintln((uint32_t)ptr); 
    memset(buffer, 0, sizeof(buffer));

    /* Get the text value of button component [the value is string type]. */
    btn->getText(buffer, sizeof(buffer));
    
    number = atoi(buffer);
    number += 1;

    memset(buffer, 0, sizeof(buffer));
    itoa(number, buffer, 10);

    /* Set the text value of button component [the value is string type]. */
    btn->setText(buffer);
}

void setup(void)
{    
    /* Set the baudrate which is for debug and communicate with Nextion screen. */
    nexInit();

    /* Register the pop event callback function of the current button component. */
    b0.attachPop(b0PopCallback, &b0);
    
    dbSerialPrintln("setup done"); 
}

void loop(void)
{   
    /*
     * When a pop or push event occured every time,
     * the corresponding component[right page id and component id] in touch event list will be asked.
     */
    nexLoop(nex_listen_list);
}
```

jyberg, 2019-02-27:
Hi, Esp8266 has only one HW Serial port, and most probaply you must use that for Nextion communication (Software serial is not working well or some times not at all). So do not activate dbSerial, because it must use same Serial port, and all messages that are writen to that is seen by Nextion display, and most probably you editor console. Same for any other Serial.Print... calls. that can mess up the functionality. I have used Serial.Print also for debug but then must be carefull because it can mess up Nextion serial communication. If you have debug console connected you should see command that are send to Nextion when same Serial TX is used, but Nextion responces are not seen (RX). These examples not care about nexInit return code, you can send messages to Nextion via Serial communication if your Serial is configured to 9600 speed.

jyberg, 2019-02-27:
One addition info that come to m mind. How you powered the nextion and board? Do you take Nextion power from board 5vcc pin? power requlator on boar is not nessessary good enough and also inside usb-> serial cable connector has power requlator that may effect.. So use separate power for Nextion and remember to connect nextion and board to same ground (gnd). I do so that I connect Usb-> serial cable to NodeMcu/Esp8266 and use another USB cable that I connect to power extension ship where I can then connect Nextion power (+5v) and gnd, and Boar 5vcc in and gnd. This ensures enough current for Nextion and board, Nextions and board is quite sensitive for power quality and if quality is not good enough RX signal can round up or it max malue is not high enough and it can cause problems...

In general Nextion communication is wery simple and not nessessary even need any special library Serial port is enough with 9600 bit rate. and commans are send as characters with out null character anc commans ends fit thee 0xFF bytes. Nextion send data back with starting one byte identification, optional data + optional thee 0xFF bytes debending the case as you can see fron Nextion Instruction Set documenation. So problem must be in serial communication settings/usage or communication quality (usually power problem)

pfeno, 2019-02-27:
@jyberg 
I comment the dbSerialdebug line in `NexConfig.h`.
I try with `SoftSerial`, `NeoSWSerial` and also `AltSoftSerial` and same result.

I totaly agree with you when you write :
> In general Nextion communication is wery simple and not nessessary even need any special library Serial port is enough with 9600 bit rate. and commans are send as characters with out null character [...]

And each (ESP & Nextion) are connect to a [Aukru Raspberry Pi 3 Alimentation 5V 3000mA avec Interrupteur pour Raspberry Pi 3 Model B+, Pi 3 modèle B,Pi 2](https://www.amazon.fr/gp/product/B01M58O9M9/ref=ppx_yo_dt_b_asin_title_o03_s01)

pfeno, 2019-03-02:
> One addition info that come to m mind. How you powered the nextion and board? Do you take Nextion power from board 5vcc pin? power regulator on boar is not necessary good enough and also inside usb-> serial cable connector has power regulator that may effect. So use separate power for Nextion and remember to connect Nextion and board to same ground (GND). I do so that I connect USB -> serial cable to NodeMcu/Esp8266 and use another USB cable that I connect to power extension ship where I can then connect Nextion power (+5v) and GND, and Board 5vcc in and GND. This ensures enough current for Nextion and board, Nextions and board is quite sensitive for power quality and if quality is not good enough RX signal can round up or it max value is not high enough and it can cause problems...
> 
> So problem must be in serial communication settings/usage or communication quality (usually **power problem**)

I test yesterday and it's correct. I connect my ESP and Nextion with 2 USB cable on same laptop ... and all work perfectly and smoothly. When I switch one of the two, what a mess!
I'm not an electronics engineer so it's like magic for me ;) 

### So to resume, I need the same GND for ESP and Nextion 

Best regards

closed 2019-03-07 by pfeno

Connect-and-Exchange, 2020-06-19:
Okay - i'm going through the code of the library and i find all kind of things i don't like:

```

void nexLoop(NexTouch *nex_listen_list[])
{
    static uint8_t __buffer[10];
    
    uint16_t i;
    uint8_t c;  
    
    while (nexSerial.available() > 0)
    {   
        **delay(10);**
        c = nexSerial.read();
        
        if (NEX_RET_EVENT_TOUCH_HEAD == c)
        {
```

This means that when i have the following list of events:
```
NexTouch* nex_listen_list[]
{
    &p0b0,
    &p1b0,
    &p0m0,
    &p0m1,
    &p0m2,
    &p0m3,
    &p0m4,
    &p0m5,
    &p0m6,
    &p0m7,
    &p2b0,
    &p0b1,
    &p3b0,
    &p3b1,
    &p3b2,
    &p3b3,
    &p3b4,
    &p3b5,
    &p3b6,
    &p3b7,
    &p3b8,
    &p3b9,
    &p3b11,
    &p3p1,
    &p3m0,
    &p3m1,
    &p3m2
};
```
it delays for 200+ ms and becomes a blocking call. 

```

        delay(10);
        c = nexSerial.read();
        
        if (NEX_RET_EVENT_TOUCH_HEAD == c)
        {
            if (nexSerial.available() >= 6)
```

and you are checking for a single char; therefore why wait and just continue and process on the next loop.

Also i removed the:

```

bool nexInit(void)
{
    bool ret1 = false;
    bool ret2 = false;
    
`    dbSerialBegin(9600); <--`
    nexSerial.begin(9600);
```

the dbserialBegin function seems for an ESP32 not very practical. This call is almost always done as the first line during Setup() of the sketch. It seems that is also the root cause of most of the problems though i'm not 100% sure yet. 

i ran into this issue when during setup the Serial.Begin(115200) was called and once the NetxtionDisplayClass got the Init() request (where the NexInit() ) takes place the serial broke down again. 

As a principle; the Nextion class should only manipulate elements which are directly related to Nextion and the dbSerialBegin is not as it is a component that is to be consumed by the user.