itead/ITEADLIB_Arduino_Nextion #70: how to hide a nextion button with arduino code

raulcq01, 2019-03-07:
please help!
i need to hide and show a button in a nextion display at runnig time, how can i do?
thanks a lot

bdelapen, 2019-12-05:
I'm in the same situation, needing to hide a button based on a specific status at run time. I got here first and thought that I wished there was an answer. I have been tinkering a little with this and have not found a direct way of either hiding/disabling the button or just moving it off the screen (this can be done at design time but not at run time as far as I have been able to test). 

Here is what I plan on doing. I am using buttons with "crop images". I have created a 3rd version of the button image with a grayed out button (though you could create one without any button at all). You can programatically change the button image so I plan on setting the button state (both picc and picc2) to the 3rd image and when re-enabling the button setting them back to the original. You have to keep track of what the status of the button is within your arduino code as the button will still send events when the images are in the "disabled" mode so you will have to ignore button clicks in your code.

I will report back when I have tested this approach in arduino, so far I have only tested it in the debug window of the Nextion editor.

bdelapen, 2019-12-07:
After some more tinkering I found that it is actually pretty straight forward to hide an element:

```
  Serial1.print("vis b3,0");
  Serial1.write(0xff);
  Serial1.write(0xff);
  Serial1.write(0xff);
```
in this case b3 = element to hide or show, 0 to hide, 1 to show.

Remember that Nextion is very particular about whitespace, leave no whitespace except between the vis instruction and the element name (you can also use element ID).

scls19fr, 2019-12-08:
A more object oriented approach could be to add

```C++
class NexObject 
{
public: /* methods */
...

    bool setVisible(bool flag);
}
```

to `NexObject.h`

and

```C++
bool NexObject::setVisible(bool flag)
{
    const char *name = getObjName();
    if (!name)
    {
        return false;
    }
    String cmd = String("vis ");
    cmd += name;
    cmd += ","
    if (flag) {
        cmd += "1"
    } else {
        cmd += "0"
    }
    sendCommand(cmd.c_str());
    return recvRetCommandFinished();
}
```

to `NexObject.cpp`

PS : be aware that it's untested... so you probably need some fixes.

sky022, 2020-05-31:
Code from @scls19fr worked well for me, with some fixes:
- few missing semi-colons
- need to add include statement for NexHardware.h at NexObject.cpp beginning
`#include "NexHardware.h"`

BMac099, 2020-09-28:
Can you tell me how I would use the above code ? I tried  "objectName.setVisible(true);" and objectName::setVisible(true); and objectName.setVisible() = true; with no luck .  Thank you.

sky022, 2020-09-29:
I'm using pointer objects to some text fields and radiobuttons, and it works well for me
`text_field->setVisible(true);`