itead/ITEADLIB_Arduino_Nextion #73: reading current page id

Deltaplc, 2019-04-15:
Good morning
Excuse me for my English (I use a translator)
You could help me understand how to read and use the current page id, and then allow you to send a temperature probe only when it is on a specific page.
Thank you

scropion86, 2019-04-17:
you have two option

- to set a global variable in Nextion screen editor and periodically update it with the Temp and once page initialized set that value to text object in nextion.
- call back void in arduino to update the text value when the page selected.

Deltaplc, 2019-04-17:
Hello,
Thank you very much for the answer, I think I understood the method, I will try to put your advice into practice.
Given that new for the programming of arduino and nextion could you give me an example?
Thanks again!

scropion86, 2019-04-17:
this is how i am doing it , i am sure this maybe not the best way to do it , but it works for me

- you  need to declare the page

```
/*
 * Declare Nextion pages [page id:0,component id:1, component name: "bt0"]. 
 */
NexPage page0    = NexPage(0, 0, "page0");  //Main page outside temp and humd
```

- and add to it to your setup

`    page0.attachPop(page0PopCallback);
`

- and add the page to the touch event listen

```
NexTouch *nex_listen_list[] = 
{
    &page0,
    NULL
};
```

- and here you update the data when selected page active

```
void page0PopCallback(void *ptr)
{
              terminal.println(F("page 0 selected"));
              terminal.println(F("---------------"));
              terminal.flush();
\\ add your probe code here which updates your nextion text value

}
```

- but you must add this in Nextion editor for page pre- initializing where you replace the X with the page index

`printh 65 0X 00 00 FF FF FF`

Deltaplc, 2019-04-17:
Hello,
Thank you so much for your help!
I added a timer in nextion that sends ip page every second so it reads the temperature variations of the DS18B20 temperature sensors.
Now everything works fine !!
Before if I sent the signal (DS18B20) also to the other pages where there was no display of them, arduino did not read the touch buttons on other pages nextion, I think due to problems with serial conflicts.
The arduino serial monitor continued to write "recvRetCommandFinished err".

Thanks again!

scropion86, 2019-04-18:
why not using global variable (in nextion side)so you can update it regardless which page is currently appearing the display.
glad you solved your issue , again there is many ways to achieve your object you select what suits your situation

MoleEng, 2019-05-22:
Add in NexHardware.cpp

`#define NEX_RET_CURRENT_PAGE_ID_HEAD        (0x66) 

/**
 * Return current page id.   
 *  
 * @param pageId - output parameter,to save page id.  
 * 
 * @retval true - success. 
 * @retval false - failed. 
 * Delay was 50. changed to 10.
 */
bool sendCurrentPageId(uint8_t* pageId)
{
 
    bool ret = false;
    uint8_t temp[5] = {0};
 
    if (!pageId)
    {
        goto __return;
    }
    sendCommand("sendme");
    delay(10);  //
    nexSerial.setTimeout(500);
    if (sizeof(temp) != nexSerial.readBytes((char *)temp, sizeof(temp)))
    {
        goto __return;
    }
 
    if (temp[0] == NEX_RET_CURRENT_PAGE_ID_HEAD
    && temp[2] == 0xFF
    && temp[3] == 0xFF
    && temp[4] == 0xFF
    )
    {
        *pageId = temp[1];
        ret = true;
    }
 
    __return:
 
    if (ret) 
    {
        dbSerial.print("recvPageId :");
        dbSerial.println(*pageId);
    }
    else
    {
       dbSerial.println("recvPageId err");
    }
 
    return ret;
 
}`

Add in Nexhardware.h
`bool sendCurrentPageId(uint8_t* pageId);`

In your loop call -- 

`uint8_t pagenumber;
sendCurrentPageId(&pagenumber);`

Deltaplc, 2019-05-23:
Thank you very much for your answer!
A good solution, now I have to think about how to call the 'sendCurrentPageId (& pagenumber);'
Only when I change page in nextion to avoid clogging the serial port and block other communications between arduino and nextion.
Thanks again
Greetings

VictorMarinus, 2023-07-18:
@MoleEng Thank you!! It works.
I have been struggling two evenings and didn't want to give up. Thanks to your code it works. (I did have to change a bit as it didn't want to compile.

So here is what I added to Nexhardware.cpp
`#define NEX_RET_CURRENT_PAGE_ID_HEAD (0x66)
/*
 * Return current page id.
 *
 * @param pageId - output parameter,to save page id.
 *
 * @RetVal true - success.
 * @RetVal false - failed.
 *Delay was 50. changed to 10.
 */
bool sendCurrentPageId(uint8_t *pageId)
{

	bool ret = false;
	uint8_t temp[5] = {0};

	if (!pageId)
	{
		goto __return;
	}

	sendCommand("sendme");
	delay(10); //
	nexSerial.setTimeout(500);
	if (sizeof(temp) != nexSerial.readBytes((char *)temp, sizeof(temp)))
	{
		goto __return;
	}

	if (temp[0] == NEX_RET_CURRENT_PAGE_ID_HEAD
		&& temp[2] == 0xFF
		&& temp[3] == 0xFF
		&& temp[4] == 0xFF
		)
	{
		*pageId = temp[1];
		ret = true;
	}

__return:

	if (ret)
	{
		dbSerial.print("recvPageId :");
		dbSerial.println(*pageId);
	}
	else
	{
		dbSerial.println("recvPageId err");
	}

	return ret;
}
`
And I added this to Nexhardware.h
`
bool sendCurrentPageId(uint8_t *pageId);
`
Only difference is the * in the "bool sendCurrentPageId(uint8_t *pageId)"

Thanks again!