itead/ITEADLIB_Arduino_Nextion #75: Unbelievably bad library

RalfNDUS, 2019-07-24:
It is unbelievable, that such a good product concept will be destroyed by one of worst library implementations I have ever seen.

I like the idea of a HMI device and it´s look brilliant on paper. That was the reason why I bought a Nextion NX4832T035-Display in order to get a great solution for my Wemos D1 mini-based projects.

I tried to used the examples in order to get familar with the programming concept. Surprisingly the display didn´t accept the provided tft-files because they are "too old". After I build my own GUI with the editor I get the next surprise because the example sketches can´t be compiled because of library issues.

Is it really a big problem to provide a library with working example files? Is there any kind of software quality check when a new release will be published? I tried both libraries, the latest and the latest stable, which makes no difference.

Hopefully another company will take the display concept and sell their products with the minimum level of quality check which a customer expects.

Trottero, 2019-09-18:
You should be able to use the Nextion instruction set by writing over Serial as is detailed [here](https://nextion.tech/instruction-set/) :). If there's an issue with the examples, fix it and create a PR, I'm sure they would be happy to accept one.

noeldum, 2021-03-25:
Did you try NeoNextion from DanNixon?
https://github.com/DanNixon/NeoNextion