itead/ITEADLIB_Arduino_Nextion #79: Will library be updated for Intelligent series?

newace150, 2019-11-04:
Hi,
am surely wrong, but is theer an uodated version released inclusive of new Intelligent series objects?
Thank you

newace150, 2020-02-09:
I guess answer is non and that I'll have to continue sending serial strings.
Appreciate help

Cheers

bryn51, 2020-05-26:
Now its May 2020 and I have to ask the same question.
Why would I buy the Intelligent series of display if there is no explicit library support for it, and I have to write my own  by extending existing functions/methods ??
Its interesting that the github readme.md file does not include the same text found on itead site vis:
"Newer components and Intelligent Series support has not been added."
Surely its about time the extra support was added in the library ?