itead/ITEADLIB_Arduino_Nextion #86: Library update for Arduino Nano 33 IoT and  33 BLE?

Bjerg13, 2020-02-06:
Hallo,
The new Arduino Nano’s (Every, 33 IoT, 33 BLE and 33 BLE Sense) are using new MCU’s and are not compadeble with the current ITEADLIB_Arduino_Nextion library. Will there be an update anytime soon?

Thanks

stef-ladefense, 2020-02-09:
il faut toujours avoir de l'espoir !
mais il faudrait pour ça que Itead se sortent les doigts du cul... pour réécrire complètement une vrai librairie et non pas reprendre la merde qu'ils ont chier actuellement.

MCR68, 2022-04-21:
I just had same problem with Nano 33 BLE and solved in this way:
Modify  NexConfig.h in ...Arduino\libraries\ITEADLIB_Arduino_Nextion directory,  row 37 "#define nexSerial Serial    //Set "Serial" for Arduino Uno "
to "#define nexSerial Serial**1**    //Set "Serial" for Arduino Uno,  Set "Serial1" for Arduino Nano BLE"
Use Tx1/Rx0 of Nano 33 BLE to communicate with Nextion Display.
in Arduino void setup()
{
....your code here..
    Serial.begin(115200);                                            // Start serial comunication at baud=115200 to Serial Monitor
    Serial1.begin(115200);                                           // Start serial comunication at baud=115200 to display Nextion

//  nexInit();                                                       //DON'T CALL nexInit(); !!! Comment it

    page0.attachPush(page0PushCallback);   //Page press event every time the GUI enter on the page 0
    page1.attachPush(page1PushCallback);   //Page press event every time the GUI enter on the page 1
    page2.attachPush(page2PushCallback);   //Page press event every time the GUI enter on the page 2
    page3.attachPush(page3PushCallback);   //Page press event every time the GUI enter on the page 3
....
}

then in

ademmler, 2022-09-14:
Hi there,

I ra into the same issue with Nextion 10'' and Arduino Nano 33 IoT . I followed your advice - thx for that.
When I run my project with the Arduino IDE open it works perfectly.
but when I close the IDE and power the Arduino from an external power supply via USB the display 
and the Nano boots/appears but nothing happens.

Seems like the communication is "bridged" within Arduino IDE or using USB ...
Can somebody help here pls?

muldereric, 2022-09-14:
I don't know anything about your setup or ino file, so this is a wild guess. Make sure you connect your display to a free physical serial port (pref not a software emulated serial port) and not the one shared by the arduino for programming and monitoring.

ademmler, 2022-09-15:
Hello,

ofcourse I use a free serial port. Otherwise it wound work if the IDE is open also ;-)
And of course I have followed Nextion advice how to change Nex.config
/**
 * Define nexSerial for communicate with Nextion touch panel. 
 */
//#define nexSerial Serial2
#define nexSerial Serial1 //Set "Serial" for Arduino Uno, Set "Serial1" for Arduino Nano BLE"

And as I wrote, the hardware is Nano 33 IoT. 
A sample sketch you can find here:
ITEADLIB_Arduino_Nextion-0.9.0/examples/CompSlider

**However I found a solution - while the reason for this problem is still unknown.
I had to include  #include <NexSlider.h> explicitly. Even I have #include <Nextion.h> in my ino.**
Seems linke the linker/compiler does something wrong here and it might be an Arduino IDE issue also ...