itead/ITEADLIB_Arduino_Nextion #111: Arduino Giga Problem

donemuhendislik, 2023-04-01:
Hi All,
Hi all, we try to develop a project by using Arduino Giga R1 WiFi. According to our project, we needed to add a screen Nextion (NX8048P050-011). According to tutorials, we have some options for this as using Nextion.h and just SoftwareSerial.h. Our problem is while we download and include your library, we have an error like this : #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors

How can we solve this problem ? Or you can advice another solution and/or library ?

donemuhendislik, 2023-04-01:
Full Error List is here :

C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:362:2: error: #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors
 #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors
  ^~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp: In member function 'bool SoftwareSerial::listen()':
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:418:23: error: 'SREG' was not declared in this scope
     uint8_t oldSREG = SREG;
                       ^~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:418:23: note: suggested alternative: 'SET'
     uint8_t oldSREG = SREG;
                       ^~~~
                       SET
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:419:5: error: 'cli' was not declared in this scope
     cli();
     ^~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX(uint8_t)':
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:593:22: error: 'digitalPinToBitMask' was not declared in this scope
   _transmitBitMask = digitalPinToBitMask(tx);
                      ^~~~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:593:22: note: suggested alternative: 'digitalPinToPinName'
   _transmitBitMask = digitalPinToBitMask(tx);
                      ^~~~~~~~~~~~~~~~~~~
                      digitalPinToPinName
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:595:27: error: 'portOutputRegister' was not declared in this scope
   _transmitPortRegister = portOutputRegister(port);
                           ^~~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:595:27: note: suggested alternative: '_transmitPortRegister'
   _transmitPortRegister = portOutputRegister(port);
                           ^~~~~~~~~~~~~~~~~~
                           _transmitPortRegister
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setRX(uint8_t)':
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:604:21: error: 'digitalPinToBitMask' was not declared in this scope
   _receiveBitMask = digitalPinToBitMask(rx);
                     ^~~~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:604:21: note: suggested alternative: 'digitalPinToPinName'
   _receiveBitMask = digitalPinToBitMask(rx);
                     ^~~~~~~~~~~~~~~~~~~
                     digitalPinToPinName
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:606:26: error: 'portInputRegister' was not declared in this scope
   _receivePortRegister = portInputRegister(port);
                          ^~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int)':
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:617:31: error: 'table' was not declared in this scope
   for (unsigned i=0; i<sizeof(table)/sizeof(table[0]); ++i)
                               ^~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:617:31: note: suggested alternative: 'tanl'
   for (unsigned i=0; i<sizeof(table)/sizeof(table[0]); ++i)
                               ^~~~~
                               tanl
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:633:9: error: 'digitalPinToPCICR' was not declared in this scope
     if (digitalPinToPCICR(_receivePin))
         ^~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:633:9: note: suggested alternative: 'digitalPinToPort'
     if (digitalPinToPCICR(_receivePin))
         ^~~~~~~~~~~~~~~~~
         digitalPinToPort
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:635:46: error: 'digitalPinToPCICRbit' was not declared in this scope
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:635:46: note: suggested alternative: 'digitalPinToPort'
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                              ^~~~~~~~~~~~~~~~~~~~
                                              digitalPinToPort
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:635:42: error: '_BV' was not declared in this scope
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                          ^~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:635:42: note: suggested alternative: '_B'
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                          ^~~
[... 41 lines cut ...]
[... cut ...]
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:735:21: error: 'SREG' was not declared in this scope
   uint8_t oldSREG = SREG;
                     ^~~~
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:735:21: note: suggested alternative: 'SET'
   uint8_t oldSREG = SREG;
                     ^~~~
                     SET
C:\Users\softw\OneDrive\Documents\Arduino\libraries\SoftwareSerial\SoftwareSerial.cpp:736:3: error: 'cli' was not declared in this scope
   cli();
   ^~~

exit status 1

Compilation error: exit status 1

Canttax, 2023-10-01:
Hello, I saw that you had found a solution to connect and exchange values between a Nextion display and a GIGA R1 Wifi arduin.

Would it be possible to have a short explanation of how it works?
For a project I'm trying to send values (e.g. speed, position relative to north...) but I can't get them to display on the screen.
Have you used a voltage booster to raise the arduino's logic level from 3.3V to 5V? 
Did you use the nextion library?
If you have the time, would it be possible to have a small example of the "typical" code for this kind of data?

Thank you in advance.
Best regards

donemuhendislik, 2023-10-01:
> Hello, I saw that you had found a solution to connect and exchange values between a Nextion display and a GIGA R1 Wifi arduin.
>
> Would it be possible to have a short explanation of how it works? For [...]

Hi @Canttax ,

Yes, I solved the problem. No one typed anything or advised a solution. After I wrote this help message, I found the solution.

First of all, just use UART communication ports. There are 4 couples of UART port. (You can find ports and explanations on original documents. Please check Arduino Giga webpage.)

For your second question, no. I don't use any library. I just used message sending / reading on UART. For clue, don't forget EOF endings 3 times on Serial2.print. (ie : Serial2.print("0x00");Serial2.print("0x00");Serial2.print("0x00");)

I could not share any codes because I used this for a customer project.

Good luck 👍🏻